rust-osdev/bootloader #447: APIC init failed on UEFI, but work on BIOS

xuanplus, 2024-07-10:
I have added ioapic to my code, and keyboard is working.

When I select boot with UEFI, there is a panic on 0x200FEE0320 or something like this, I forget it.

So an error happens when I init Timer IRQ.

Forgive my bad grammar.

What should I do for this?

Freax13, 2024-07-10:
The page fault happens when your kernel tries to access the local APIC at `0xfee0_0000`. This currently fails because the physical memory mapping only covers [all the memory regions](https://github.com/rust-osdev/bootloader/blob/bf950a4bee82764dac17b5e18f5c82f491c9799e/common/src/lib.rs#L351-L353) we get from UEFI and if the system doesn't have a lot of memory, the mapping won't cover `0xfee0_0000`. Passing `-m 4G` to QEMU forces the physical memory mapping to be large enough to cover the local APIC and makes the access succeed.

I'd be okay with making the physical memory mapping always cover at least the first 4 GiB of physical memory. Feel free to open a PR if you want to, otherwise I'd also be happy to implement this.

xuanplus, 2024-07-10:
Thanks for your help!

Althought it maybe temporary, it works perfectly.

By the way, I'm true noob in os programming, so open the pr please, and I can learn more from it.

phil-opp, 2024-07-11:
> I'd be okay with making the physical memory mapping always cover at least the first 4 GiB of physical memory. Feel free to open a PR if you want to, otherwise I'd also be happy to implement this.

Sounds like a good idea!

PR #448 (2024-07-11): always cover at least the first 4 GiB of physical memory
Always cover at least the first 4 GiB of physical memory. That area contains useful MMIO regions (local APIC, I/O APIC, PCI bars) that we want to make accessible to the kernel even if no DRAM exists >4GiB.

Closes #447
Cc @xuanplus
closed 2024-07-11 by phil-opp