rust-osdev/bootloader #449: Nightly Cargo.lock does not exist.

bingmatv, 2024-08-08:
~~~
  error: failed to compile `bootloader-x86_64-bios-stage-3 v0.11.7`, intermediate artifacts can be found at `/tmp/cargo-installL6j4n0`.
  To reuse those artifacts with a future compilation, set the environment variable `CARGO_TARGET_DIR` to that path.

  Caused by:
    "/home/wzh/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/Cargo.lock" does not exist, unable to build with the standard library, try:
            rustup component add rust-src --toolchain nightly-x86_64-unknown-linux-gnu
  thread 'main' panicked at /home/wzh/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bootloader-0.11.7/build.rs:286:9:
  failed to build bios stage-3
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
   Downloading crates ...
  error: failed to compile `bootloader-x86_64-bios-boot-sector v0.11.7`, intermediate artifacts can be found at `/tmp/cargo-installGCZCyV`.
  To reuse those artifacts with a future compilation, set the environment variable `CARGO_TARGET_DIR` to that path.

  Caused by:
    "/home/wzh/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/Cargo.lock" does not exist, unable to build with the standard library, try:
            rustup component add rust-src --toolchain nightly-x86_64-unknown-linux-gnu
    Downloaded byteorder v1.4.3
    Downloaded bit_field v0.10.2
    Downloaded autocfg v1.1.0
    Downloaded hash32 v0.2.1
    Downloaded heapless v0.7.16
    Downloaded scopeguard v1.1.0
    Downloaded ptr_meta v0.2.0
    Downloaded ptr_meta_derive v0.2.0
    Downloaded rand_core v0.6.4
    Downloaded stable_deref_trait v1.2.0
    Downloaded xmas-elf v0.8.0
    Downloaded volatile v0.4.6
    Downloaded ucs2 v0.3.2
    Downloaded uefi-macros v0.11.0
    Downloaded rand_hc v0.3.2
    Downloaded quote v1.0.29
    Downloaded log v0.4.19
    Downloaded lock_api v0.4.10
    Downloaded semver v1.0.17
    Downloaded spin v0.9.8
    Downloaded rustc_version v0.4.0
    Downloaded serde_derive v1.0.166
    Downloaded unicode-ident v1.0.10
    Downloaded spinning_top v0.2.5
    Downloaded ryu v1.0.14
    Downloaded proc-macro2 v1.0.63
    Downloaded serde v1.0.166
    Downloaded rand v0.8.5
    Downloaded raw-cpuid v10.7.0
    Downloaded uefi v0.20.0
    Downloaded syn v2.0.23
    Downloaded conquer-util v0.3.0
    Downloaded conquer-once v0.3.2
    Downloaded bootloader-x86_64-common v0.11.7
    Downloaded syn v1.0.109
    Downloaded zero v0.1.3
    Downloaded usize_conversions v0.2.0
    Downloaded uart_16550 v0.2.18
    Downloaded rustversion v1.0.13
    Downloaded serde-json-core v0.5.0
    Downloaded x86_64 v0.14.10
    Downloaded noto-sans-mono-bitmap v0.2.0
  error: failed to compile `bootloader-x86_64-uefi v0.11.7`, intermediate artifacts can be found at `/tmp/cargo-installXMseyY`.
  To reuse those artifacts with a future compilation, set the environment variable `CARGO_TARGET_DIR` to that path.

  Caused by:
    "/home/wzh/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/Cargo.lock" does not exist, unable to build with the standard library, try:
            rustup component add rust-src --toolchain nightly-x86_64-unknown-linux-gnu
   Downloading crates ...
    Downloaded mbr-nostd v0.1.0
  error: failed to compile `bootloader-x86_64-bios-stage-2 v0.11.7`, intermediate artifacts can be found at `/tmp/cargo-installUYlz3V`.
  To reuse those artifacts with a future compilation, set the environment variable `CARGO_TARGET_DIR` to that path.

  Caused by:
    "/home/wzh/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/Cargo.lock" does not exist, unable to build with the standard library, try:
            rustup component add rust-src --toolchain nightly-x86_64-unknown-linux-gnu
   Downloading crates ...
    Downloaded rsdp v2.0.0
  error: failed to compile `bootloader-x86_64-bios-stage-4 v0.11.7`, intermediate artifacts can be found at `/tmp/cargo-installXIcHkD`.
  To reuse those artifacts with a future compilation, set the environment variable `CARGO_TARGET_DIR` to that path.

  Caused by:
    "/home/wzh/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/Cargo.lock" does not exist, unable to build with the standard library, try:
            rustup component add rust-src --toolchain nightly-x86_64-unknown-linux-gnu
~~~

bjorn3, 2024-08-08:
`-Zbuild-std` is broken with the latest nightly. Try using a nightly from a couple of days ago. `-Zbuild-std` should be fixed within a couple of days.

bingmatv, 2024-08-08:
@bjorn3 There's the same issue on ARM64 device. I tried
~~~
cargo generate-lockfile
~~~
In the project folder, or
~~~
find ~/.rustup/toolchains/nightly-aarch64-unknown-linux-gnu|grep Cargo.toml
~~~
will find some toml files on ARM64 host. Still run
~~~
cd ~/.rustup/toolchains/nightly-aarch64-unknown-linux-gnu/lib/rustlib/src/rust
cargo generate-lockfile
~~~
copy the generated lock file to the expected folder, and works on ARM64 host. It might work on x86_64 hosts. If there aren't any obvious differences between ARM64 and x86 environments, it may work on x86.

bingmatv, 2024-08-09:
> If there aren't any obvious differences between ARM64 and x86 environments, it may work on x86.

I tried it on x86_64, it also works. If the host is x86_64, it also has
~~~
lib/rustlib/src/rust
~~~
in the toolchain folder, copy the generated lockfile into that folder fixed this.

Freax13, 2024-08-12:
This issue appears to be [fixed](https://github.com/rust-osdev/bootloader/actions/runs/10329385717) as of two days ago.