rust-osdev/bootloader #451: using braces for entry_point! saves 1 byte

bingmatv, 2024-08-09:
The 3rd edition of the blog https://github.com/phil-opp/blog_os/blob/edition-3/blog/content/edition-3/posts/02-booting/index.md looks like
~~~
bootloader_api::entry_point!(kernel_main);
~~~
Using braces like
~~~
bootloader_api::entry_point!{kernel_main}
~~~
Doesn't need the lasting semicolon so it saves 1 byte.

bjorn3, 2024-08-09:
Using braces is less clear IMO. I don't see any reason why you would want to save a single byte in the source code.

Freax13, 2024-08-13:
The different bracket styles are usually used for different things: Normal parenthesis are used for function-call-like macros, curly braces are usually used when the body contains decl- or statements-like content, and brackets are used for array-like things. We should stick to those conventions: In this case, the macro body doesn't contain decl- or statement-like tokens and doesn't have anything to do with collections of any sort. Considering that changing this would only save a single byte,IMO  we should prioritize clarity.

phil-opp, 2024-08-21:
As bjorn3 noted, the compiled binaries are not affected by brace styles, variable names, newlines, etc. For distributing the source files, you can compress them (e.g. zip them). So there is no real advantage of saving a few bytes in the source file.

Because of this, we can freely optimize the source file for readability, as Freax13 mentioned.

closed 2024-08-21 by phil-opp