rust-osdev/bootloader #452: add TextBuffer

bingmatv, 2024-08-13:
There's FrameBuffer but no TextBuffer.

bjorn3, 2024-08-13:
If you boot using BIOS, you can use VGA text mode to directly feed text to the GPU. Bowever in the case of UEFI, GOP (the UEFI replacement for the VGA and VESA device interfaces) only supports setting up framebuffers. Because of this the kernel has to turn text into pixels itself. For consistency the bootloader crate only provides a framebuffer interface, even on BIOS where it could technically provide a text mode interface.

bingmatv, 2024-11-10:
> (the UEFI replacement for the VGA and VESA device interfaces) only supports setting up framebuffers

How to load and parse existing fonts developed by people? E.g, to copy the fonts from Host OS.

bjorn3, 2024-11-10:
Parsing and rendering of regular vector fonts is very complex (and slow inside a kernel due to needing to use soft floats there) and almost certainly not what you want in the kernel. (it has lead to countless of kernel exploits in the case of windows for example) Rather you likely want to use a pre-rasterized font instead. There are several crates that contain pre-rasterized fonts like noto_sans_mono_bitmap (containing Noto Sans Mono) Or you can use the `mono_font` module of the [embedded-graphics](https://docs.rs/embedded-graphics/latest/embedded_graphics/mono_font/index.html) crate.

bingmatv, 2024-11-10:
> lead to countless of kernel exploits in the case of windows for example

Why vectors and floats lead to kernel exploits?