rust-osdev/bootloader #454: new issues in blog os repo

bingmatv, 2024-08-22:
https://github.com/phil-opp/blog_os/issues E.g., an issue says there may be a typo "444" in the 3rd edition.

Freax13, 2024-08-22:
That's unrelated to this repo, so I'm closing this.

A small piece of advice: If you notice a typo and want to get it fixed, opening a PR is probably a better strategy than opening an issue. We all lead busy lifes and just pushing the merge button is a lot less effort on the maintainer's part rather than fixing the typo themselves (fixing it themselves probably requires them to at least check out the repo, fetch the latest changes, stash WIP changes, checking out the latest main branch, create a new branch, fix those typos and then creating a pull request). Fixing typos yourself is also a nice and pretty easy way to start contributing to a project and get more involved :smiley:

closed 2024-08-22 by Freax13

bingmatv, 2024-08-24:
> unrelated to this repo

Why did phil-opp posted something in https://github.com/rust-osdev/bootloader/issues/299 ?

phil-opp, 2024-08-25:
Sorry for not getting to your issues earlier! Feel free to ping me in the issue thread or on our [zulip chat](https://rust-osdev.zulipchat.com/) if you don't get a reply for a long time. However, please don't open additional issues like this one in other, unrelated repositories. It just creates additional work for us.