rust-osdev/bootloader #455: looks possible to use std::fs::File::write_all

bingmatv, 2024-08-22:
Rust has [write_all()](https://doc.rust-lang.org/stable/std/io/trait.Write.html#method.write_all) in std::fs::File, it may be able to write certain numbers (machine code) into a file. Can it replace some Assembly and remove some Nightly feature usages in this repo?

Freax13, 2024-08-22:
Can you elaborate on what specific files you want to write to?

> Can it replace some Assembly

I don't think we'll want to replace assembly with straight-up machine code. Machine code is unreadable and difficult to maintain whereas assembly isn't (though we try to use as much Rust as possible).

> and remove some Nightly feature usages in this repo?

AFAICT we only use [one](https://github.com/rust-osdev/bootloader/blob/96f58f2d78cb9d34e6f963aa14f0307bcc953fe8/common/src/lib.rs#L2) nightly rustc feature directly and probably some of them indirectly through dependencies. We might be able to cut down on some of these, but we won't be able to remove them completely until binary dependencies are stabilized in cargo (I haven't seen much progress on that lately). That said we haven't had many problems with nightly breakages in the past few months, so I don't see a strong need to cut down on nightly features.

bingmatv, 2024-08-22:
> Can you elaborate on what specific files you want to write to?

To make bootable images for BIOS and UEFI.

Freax13, 2024-08-22:
We already make bootable BIOS and UEFI images.