rust-osdev/bootloader #462: Hang after loading first level 4 table on uefi

Colepng, 2024-10-24:
I tried my os on real hardware for the first time and found that the bootloader is hanging. I tracked the issue down to this [line](https://github.com/rust-osdev/bootloader/blob/fffa7ceb317328ed19bc0ebaa0028be152095a05/uefi/src/main.rs#L336-L339). I'm still a beginner in os development so any help is appreciated.

I am running it on my laptop which is a lenvo legion 7 which has a AMD Ryzen 7 5800H.

Freax13, 2024-10-24:
Hmm, there's not much to go on. How did you track this hang down to this line? Can you share the log output?

Colepng, 2024-10-25:
I cloned the repo locally, linked my kernel to the local version and added debug statements. I put 1 statement after the line I linked to and 1 after. Only the first 1 ran. The best photo I can add would be from my phone.

[image]

As you can see in the photo it no longer hang. Instead of only copying over the first entry of the old page table, I started copying over all entries and now it works. The panic you can see in the photo seems to be from the fact my system uses huge page table while the bootloader expects 4KiB tables. If you know, could you explain to me why copying over all the old entries would fix it.

Freax13, 2024-10-25:
The frame buffer (0xfc_1000_0000) is address is in the second pml4 entry, interesting, so the assumption in the comment was incorrect. Your fix seems reasonable. Freel free to open at PR.

PR #466 (2024-11-14): copy more PML4 entries
If there is more than 512 GiB of memory, we need to copy more than the first PML4 entry to access it. Similarly if the frame buffer is not in the address range covered by the first PML4 entry, we need to copy more entries in order to access it. We also need to mark these PML4 entries as unusable when allocating virtual addresses for the kernel because the bootloader maps some of them into its own address space.

This is a much simpler solution to #462 than #465. The downside is that we mark some of the kernel's virtual address space as unusable when it doesn't technicall [... cut]
closed 2024-11-14 by Freax13