rust-osdev/bootloader #470: What's mapped in the lower half?

mrjbom, 2024-11-01:
As the kernel of the upper half, I want nothing to be displayed in the lower half.
But tlb after loading looks like this:
```
(qemu) info tlb
000000000013b000: 000000000013b000 ----A----
000000000013c000: 000000000013c000 ---------
00000000013ac000: 00000000013ac000 ---------
...
Higher half mapped
```
I realized that 13ac000 belongs to GDT, I created my GDT and can now unmap that address.
But what is mapped in the first two addresses? Can I safely unmap them?

Freax13, 2024-11-01:
That's probably the context switch function of the boot loader:
https://github.com/rust-osdev/bootloader/blob/ea3f61a57ccd57b7ed37eeed1e53ee6cb2cfdd81/common/src/lib.rs#L235-L260
These pages contain a function that switches to the kernel page tables and jumps to its entry point. You can safely unmap it.

See also #240

mrjbom, 2024-11-01:
If it turns out not to be her, then I suppose I can safely unmap these addresses anyway? Why does the loader leave these things littered?
For example, the same GDT could be safely unmapped by bootloader.

Freax13, 2024-11-01:
> If it turns out not to be context switch function, then I suppose I can safely unmap these addresses anyway? 

I'm pretty sure it's the context switch function. If it's not the context switch function, then it depends on the memory for whether it's safe to unmap.

> Why does the bootloader leave these things mapped?

The context switch function needs to do two things:
1. Switch to the kernel's page tables.
2. Jump to the kernel's entry point.

It needs to do these steps in that order. The problem is that if we don't map the context switch function into the kernel's page tables, we can't do the second step after switching the page tables.

> For example, GDT could be safely unmapped by bootloader.

The GDT can't be safely unmapped before switching to a new one. Some instructions like `iret` read from the GDT and if the GDT is no longer mapped, this causes a page fault.

closed 2024-11-01 by mrjbom