rust-osdev/bootloader #477: Unable to compile bootloader v0.11.8

kevstewa, 2024-11-24:
I am attempting to follow the edition-3 instructions. I am currently on the [Using the DiskImageBuilder](https://github.com/phil-opp/blog_os/blob/edition-3/blog/content/edition-3/posts/02-booting/index.md#using-the-diskimagebuilder) step. Once I add the bootloader dependency I get the compiler error shown below.

I've attempted to pin the bootloader and bootloader_api crate versions to those specified in the docs, with errors resolved in later versions (proc_macro). My version of Rust is 1.85.0-nightly.

```
   Compiling bootloader v0.11.8
error: failed to run custom build command for `bootloader v0.11.8`

Caused by:
  process didn't exit successfully: `/home/kstewart/Nextcloud/Projects/Rust/os2/target/debug/build/bootloader-0fce06f805b868ea/build-script-build` (exit status: 101)
  --- stderr
      Blocking waiting for file lock on package cache
      Blocking waiting for file lock on package cache
      Updating crates.io index
       Ignored package `bootloader-x86_64-bios-boot-sector v0.11.8` is already installed, use --force to override
  warning: be sure to add `/home/kstewart/Nextcloud/Projects/Rust/os2/target/debug/build/bootloader-f7327baa02045b49/out/bin` to your PATH to be able to run the installed binaries
       Ignored package `bootloader-x86_64-bios-stage-2 v0.11.8` is already installed, use --force to override
  warning: be sure to add `/home/kstewart/Nextcloud/Projects/Rust/os2/target/debug/build/bootloader-f7327baa02045b49/out/bin` to your PATH to be able to run the installed binaries
       Ignored package `bootloader-x86_64-bios-stage-3 v0.11.8` is already installed, use --force to override
  warning: be sure to add `/home/kstewart/Nextcloud/Projects/Rust/os2/target/debug/build/bootloader-f7327baa02045b49/out/bin` to your PATH to be able to run the installed binaries
      Blocking waiting for file lock on package cache
      Updating crates.io index
    Installing bootloader-x86_64-uefi v0.11.8
      Blocking waiting for file lock on package cache
      Updating crates.io index
    Installing bootloader-x86_64-bios-stage-4 v0.11.8
      Blocking waiting for file lock on package cache
      Blocking waiting for file lock on package cache
      Updating crates.io index
      Blocking waiting for file lock on package cache
  warning: package `atomic-polyfill v0.1.11` in Cargo.lock is yanked in registry `crates-io`, consider running without --locked
  warning: package `critical-section v1.1.1` in Cargo.lock is yanked in registry `crates-io`, consider running without --locked
      Blocking waiting for file lock on package cache
  warning: package `atomic-polyfill v0.1.11` in Cargo.lock is yanked in registry `crates-io`, consider running without --locked
  warning: package `critical-section v1.1.1` in Cargo.lock is yanked in registry `crates-io`, consider running without --locked
      Updating crates.io index
      Blocking waiting for file lock on package cache
      Blocking waiting for file lock on package cache
      Updating crates.io index
      Blocking waiting for file lock on package cache
      Blocking waiting for file lock on package cache
      Blocking waiting for file lock on package cache
     Compiling compiler_builtins v0.1.138
     Compiling core v0.0.0 (/home/kstewart/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core)
     Compiling proc-macro2 v1.0.63
     Compiling quote v1.0.29
     Compiling unicode-ident v1.0.10
     Compiling autocfg v1.1.0
     Compiling semver v1.0.17
     Compiling syn v1.0.109
     Compiling compiler_builtins v0.1.138
     Compiling serde v1.0.166
     Compiling rustversion v1.0.13
     Compiling bootloader_api v0.11.8
     Compiling core v0.0.0 (/home/kstewart/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core)
     Compiling rustc_version v0.4.0
     Compiling lock_api v0.4.10
     Compiling proc-macro2 v1.0.63
     Compiling unicode-ident v1.0.10
     Compiling autocfg v1.1.0
     Compiling quote v1.0.29
     Compiling syn v2.0.23
     Compiling semver v1.0.17
     Compiling heapless v0.7.16
     Compiling serde v1.0.166
     Compiling lock_api v0.4.10
     Compiling rustversion v1.0.13
     Compiling bootloader_api v0.11.8
     Compiling rustc_version v0.4.0
     Compiling syn v2.0.23
[... 112 lines cut ...]
     --> /home/kstewart/.cargo/registry/src/index.crates.io-6f17d22bba15001f/x86_64-0.14.10/src/structures/paging/page.rs:274:14
      |
  273 | /         Step::steps_between(&start.start_address, &end.start_address)
  274 | |             .map(|steps| steps / S::SIZE as usize)
      | |             -^^^ `(usize, Option<usize>)` is not an iterator
      | |_____________|
      |
      |
      = note: the following trait bounds were not satisfied:
              `(usize, Option<usize>): Iterator`
              which is required by `&mut (usize, Option<usize>): Iterator`

  thread 'main' panicked at /home/kstewart/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bootloader-0.11.8/build.rs:95:9:
  failed to build uefi bootloader
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
     Compiling rand_hc v0.3.2
     Compiling raw-cpuid v10.7.0
  Some errors have detailed explanations: E0053, E0599.
  For more information about an error, try `rustc --explain E0053`.
     Compiling spinning_top v0.2.5
  error: could not compile `x86_64` (lib) due to 3 previous errors
  warning: build failed, waiting for other jobs to finish...
  error: failed to compile `bootloader-x86_64-bios-stage-4 v0.11.8`, intermediate artifacts can be found at `/tmp/cargo-installEC0LSG`.
  To reuse those artifacts with a future compilation, set the environment variable `CARGO_TARGET_DIR` to that path.
```

bjorn3, 2024-11-24:
This is caused by https://github.com/rust-osdev/x86_64/issues/512, which will be fixed by https://github.com/rust-osdev/x86_64/pull/513.

kevstewa, 2024-11-24:
Thanks, @bjorn3 -- closing this ticket.

closed 2024-11-24 by kevstewa