rust-osdev/bootloader #480: ISO file cannot be loaded by VirtualBox

mrjbom, 2024-12-01:
I am trying to load the ISO file I created using Virtual Box, but am encountering errors when I try to add the disk to VirtualBox. I also can't open it with WinRAR. In qemu the file loads and works fine. I haven't tried loading it on real hardware, but I don't think it will work there either.

What is wrong with it?

bjorn3, 2024-12-01:
What errors are you facing with virtualbox?

mrjbom, 2024-12-01:
@bjorn3
For BiosBoot: 
```
Could not get the storage format of the medium 'bootable_biosboot.iso' (VERR_NOT_SUPPORTED).
Result Code: VBOX_E_IPRT_ERROR (0x80bb0005)
Result Detail: -37 (0xffffffdb)
Component: MediumWrap
Interface: IMedium {7d510820-a678-4730-a862-818dcd3fbed0}
Callee: IVirtualBox {d644ad1e-c501-4fc7-9ab6-aa6d763bc540}
Callee RC: VBOX_E_OBJECT_NOT_FOUND (0x80bb0001)

```
UefiBoot stopped working in qemu(just hangs from the start, nothing is displayed). In VirtualBox the disk is added, but VirtualBox can't boot it and keeps restarting asking for a new disk.

bjorn3, 2024-12-01:
How did you attach the iso to the VM? Did you attach it as cd or as harddrive?

mrjbom, 2024-12-01:
I add the iso in Virtual Media Manager as an optical disk.

mrjbom, 2024-12-01:
@bjorn3,
VirtualBox behaves very strangely with UefiBoot ISO image, as if it does not see it, despite the settings of the virtual machine, where the disk is definitely inserted.

In general, all this seems to be some kind of incorrectly formed ISO.

I tried to do a simple check with isovfy and in the case of the UefiBoot disk it complained
```
Root at extent 9510959, 156567892 bytes
[0 0]
********Directory has unusual size
isovfy: Short read on old image
```

mrjbom, 2024-12-03:
Oops, I was stupid and thought it was creating an iso file, but it's actually a disk image. Everything's fine now.

closed 2024-12-03 by mrjbom