rust-osdev/bootloader #481: Question on second stage

boredcoder411, 2025-01-12:
I've read the through the first bootsector's code, and everything looks good with objdump: _second_stage_start label is supposed to be at ram 0x7e00 and the code calls it, but I'm trying to make my own second stage and don't understand just how the second stage is placed after the first one. I've tried assembling a program with nasm -f bin and just cat appending to the original file, but that doesn't seem to work, even if the new code is at offset 0x7e00. Am I missing something?

bjorn3, 2025-01-12:
create_mbr_disk inserts padding as necessary to write it to the right location on the disk. And the first stage then loads a fixed sector on the disk to a fixed address in ram.

boredcoder411, 2025-01-12:
So I can't just write bytes at the second sector?

bjorn3, 2025-01-12:
Looks like the second stage is actually written to the second sector: https://github.com/rust-osdev/bootloader/blob/086c248e89ec3519f77dadda6c74d193e6fa22fd/src/mbr.rs#L31 When cat'ing are you sure the first file for the boot sector is exactly 512 bytes?

boredcoder411, 2025-01-12:
Yep! Because the linker script fills with 00 and 55aa at the 511 and 512 bytes

boredcoder411, 2025-01-14:
https://github.com/boredcoder411/rust_boot
If you want to check it out, but it looks fine to me