rust-osdev/bootloader #482: Bootloader compilation fails

oblivisheee, 2025-01-16:
[image]
When I add bootloader to Cargo.toml I get such thing. How do I fix this?

Freax13, 2025-01-16:
What does your Cargo.toml looks like?

oblivisheee, 2025-01-16:
@Freax13 
> What does your Cargo.toml looks like?
Cargo.toml
```toml
[package]
name = "merizo-os"
version = "0.1.0"
edition = "2021"

[profile.dev]
panic = "abort"

[profile.release]
panic = "abort"

[target.aarch64-unknown-none.dependencies]
aarch64-cpu = "10.0.0"

[dependencies]
bootloader = "0.11.9"
cortex-a = "8.1.1"
tock-registers = "0.9.0"
uefi = { version = "0.33.0", features = ["qemu"] }
uefi-raw = "0.9.0"
```

.cargo/config.toml
```toml
[unstable]
build-std = ["core", "compiler_builtins", "alloc"]
build-std-features = ["compiler-builtins-mem"]

[build]
target = "aarch64-unknown-none"

[target.aarch64-unknown-none]
rustflags = ["-C", "link-arg=-Tconf/linker.ld", "-C", "target-cpu=cortex-a72"]
linker = "rust-lld"

```

Freax13, 2025-01-16:
Just remove the bootloader dependency. If you want to write a kernel for aarch64, you can't use this crate. You can just use the uefi crate.

oblivisheee, 2025-01-16:
@Freax13 
> Just remove the bootloader dependency. If you want to write a kernel for aarch64, you can't use this crate. You can just use the uefi crate.

I have not decided yet. x86_64 is easier to write for me because of ecosystem. 
When I use x86_64-unknown-none happens same.

[image]

Freax13, 2025-01-16:
Yeah, because you're not supposed to add the bootloader dependency to the kernel itself. Please follow [this guide](https://github.com/rust-osdev/bootloader?tab=readme-ov-file#usage).

oblivisheee, 2025-01-16:
> Yeah, because you're not supposed to add the bootloader dependency to the kernel itself. Please follow [this guide](https://github.com/rust-osdev/bootloader?tab=readme-ov-file#usage).

Thanks

closed 2025-01-16 by oblivisheee