rust-osdev/bootloader #483: Dependency `instant` is unmaintained

gaojiaqi7, 2025-01-17:
```
cargo deny check advisories
2025-01-17 02:39:06 [WARN] unable to find a config path, falling back to default config
error[unmaintained]: `instant` is unmaintained
   ┌─ /home/env/source/bootloader/Cargo.lock:60:1
   │
60 │ instant 0.1.13 registry+https://github.com/rust-lang/crates.io-index
   │ ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ unmaintained advisory detected
   │
   ├ ID: RUSTSEC-2024-0384
   ├ Advisory: https://rustsec.org/advisories/RUSTSEC-2024-0384
   ├ This crate is no longer maintained, and the author recommends using the maintained [`web-time`] crate instead.
     
     [`web-time`]: https://crates.io/crates/web-time
   ├ Solution: No safe upgrade is available!
   ├ instant v0.1.13
     └── fastrand v1.9.0
         └── futures-lite v1.13.0
             ├── async-io v1.13.0
             │   └── async-process v1.8.1
             │       └── (build) bootloader v0.11.9
             │           └── bootloader_test_runner v0.1.0
             │               └── (dev) bootloader v0.11.9 (*)
             ├── async-process v1.8.1 (*)
             └── futures-concurrency v7.6.2
                 └── (build) bootloader v0.11.9 (*)
```

bjorn3, 2025-01-17:
Updating futures-lite to 2.6 would remove the dependency on instant. Unfortunately futures-concurrency doesn't have a version which depends on futures-lite 2.6. Only async-process has an updated version available.

closed 2025-01-19 by phil-opp

gaojiaqi7, 2025-01-20:
Thanks @bjorn3 @Freax13 ! The `instant` crate is still present in the dependency tree because it is indirectly referenced by `tempfile`:

```
error[unmaintained]: `instant` is unmaintained
   ┌─ /home/env/source/bootloader/Cargo.lock:26:1
   │
26 │ instant 0.1.12 registry+https://github.com/rust-lang/crates.io-index
   │ ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ unmaintained advisory detected
   │
   ├ ID: RUSTSEC-2024-0384
   ├ Advisory: https://rustsec.org/advisories/RUSTSEC-2024-0384
   ├ This crate is no longer maintained, and the author recommends using the maintained [`web-time`] crate instead.
     
     [`web-time`]: https://crates.io/crates/web-time
   ├ Solution: No safe upgrade is available!
   ├ instant v0.1.12
     └── fastrand v1.9.0
         └── tempfile v3.6.0
             └── bootloader v0.11.9
                 └── bootloader_test_runner v0.1.0
                     └── (dev) bootloader v0.11.9 (*)
```

I have submitted a PR https://github.com/rust-osdev/bootloader/pull/485 to upgrade the version of `tempfile`, so instant will be completely removed.

Freax13, 2025-01-20:
Oops, my bad, that's embarrassing 🙈

PR #484 (2025-01-19): use threads instead of futures in build.rs
There's no inherent reason why we should use future-based concurrency in build.rs. We don't spawn many futures and we don't make use of any of the nice structures made possible by structured concurrency. The downside of using futures is that we have to pull in a bunch of dependencies. Switching to sync code and removing the dependencies resulted in a nice performance improvement (35s -> 30s). Also, it's just good practice to reduce the number of unneeded dependencies especially if some of them are unmaintained.

Closes #483