rust-osdev/bootloader #497: Feature request: Create an iso that can be chainloaded with grub

ChocolateLoverRaj, 2025-03-02:
Right now the UEFI kernel cannot load the custom kernel if I try to `loopback` and `chainload` an iso after I converted the `.img` to a ISO (by mounting and creating an ISO from it).

Freax13, 2025-03-02:
> Right now the UEFI kernel cannot load the custom kernel 

Can you be more specific about what fails when you do this? Does the bootloader start? Does it print stuff? Does GRUB fail to start it?

> if I try to `loopback` and `chainload` an iso

Please provide the config you used and think should work.

> after I converted the `.img` to a ISO (by mounting and creating an ISO from it).

It's very likely that you don't need to do this conversion step. Unless you're using CD-ROMs, you can just use the .img (e.g., by flashing it onto a USB thumbdrive).

ChocolateLoverRaj, 2025-03-02:
> Can you be more specific about what fails when you do this? Does the bootloader start? Does it print stuff? Does GRUB fail to start it?

The bootloader starts but it panics and says that it failed to load the kernel.

I changed the GRUB config since then, but basically I had an iso in the grub folder and I did 
```
loopback loop /code-runner-uefi.iso
chainload (loop)/efi/boot/bootx64.efi
```

When I did `loopback loop /code-runner-uefi.img` GRUB had an error and it said unknown file system.

I don't want to flash it onto a disk because then I would need to dedicate an entire disk specifically for it.

Freax13, 2025-03-02:
> The bootloader starts but it panics and says that it failed to load the kernel.

> When I did `loopback loop /code-runner-uefi.img` GRUB had an error and it said unknown file system.

Telling us that there are error messages and then not providing them is completely unhelpful 🙈 We can't help you if you don't tell us what's going wrong.
 
I'm not sure if this *can* work. There are two problems:
1. AFAICT UEFI has no concept of mounting an image i.e. it doesn't support loopback devices. It has RAM disks, but I don't see anything that would allow using a file to back the storage of a disk. I found a [repo](https://github.com/EHfive/uefi-toys) that implements a loopback driver and the fact they had to do that suggests to me that this is not natively supported by UEFI.
2. When grub chainloads an EFI executable, it always uses the [root device](https://git.savannah.gnu.org/cgit/grub.git/tree/grub-core/loader/efi/chainloader.c#n242) (i.e. the disk containing GRUB?) when constructing the image path for the chainloaded executable. This is a problem because the bootloader tries to locate the kernel on the same device that it thinks it was loaded from, but if GRUB just passes a completely different device, the bootloader will also look for the kernel on that wrong device. This probably explains why you saw errors about the bootloader complaining that it couldn't load the kernel. In their defense, reconstructing the correct device path is not an easy problem and it can be entirely impossible when EFI doesn't support the device (see problem 1).

I think we could possibly work around the second problem by adding support for passing the root device as a commandline argument, but I don't see a way around the first problem.

ChocolateLoverRaj, 2025-03-03:
Currently the img file generated does work with Ventoy just so you know.

Freax13, 2025-03-03:
Ventoy ships its own [loopback device driver](https://github.com/ventoy/Ventoy/tree/master/EDK2/edk2_mod/edk2-edk2-stable201911/MdeModulePkg/Application/VDiskChain), that's why they can pull that off.

I think the sad reality is that UEFI chainloading just isn't well supported by grub. You might have more luck with a different bootloader.

ChocolateLoverRaj, 2025-03-03:
I think I will just use set Ventoy to auto boot my OS and tell GRUB (or another bootloader) to chain boot Ventoy.

closed 2025-03-03 by Freax13