rust-osdev/bootloader #499: Unable to compile bin `bootloader` due to error from rust-lld

9thIvy, 2025-03-20:
I originally came to this error after trying to build blog_os as I was following it. It's worked flawlessly so far, but I got into dependency hell on my last linux install and ended up nuking it. After installing nobara 41 over 40, I've been getting an error from rust-lld when building.

I have truncated most of the cargo build and install output. Here are my steps to reproduce:

1.
```
$ git clone https://github.com/phil-opp/blog_os.git && cd blog_os
```
2.
```
$ git checkout -b post-12 origin/post-12 
```
3. 

```
$ cargo build
Compiling compiler_builtins v0.1.151
   Compiling core v0.0.0 (/home/mythril/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core)
   Compiling rustc-std-workspace-core v1.99.0 (/home/mythril/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/rustc-std-workspace-core)
   Compiling alloc v0.0.0 (/home/mythril/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/alloc)
   Compiling blog_os v0.1.0 (/tmp/blog_os)
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 10.22s
```
4.
```
$ cargo install bootimage
$ cargo install bootimage
    Updating crates.io index
  Installing bootimage v0.10.3
    Updating crates.io index
     Locking 22 packages to latest compatible versions
   Compiling bootimage v0.10.3
    Finished `release` profile [optimized] target(s) in 5.22s
  Installing /home/mythril/.cargo/bin/bootimage
  Installing /home/mythril/.cargo/bin/cargo-bootimage
   Installed package `bootimage v0.10.3` (executables `bootimage`, `cargo-bootimage`)
```
5.
```
$ cargo bootimage
WARNING: `CARGO_MANIFEST_DIR` env variable not set
Building kernel
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.04s
Building bootloader
   Compiling compiler_builtins v0.1.151
   Compiling core v0.0.0 (/home/mythril/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core)
   Compiling serde v1.0.116
   Compiling bitflags v1.2.1
   Compiling x86_64 v0.14.7
   Compiling llvm-tools v0.1.1
   Compiling toml v0.5.6
   Compiling bootloader v0.9.30 (/home/mythril/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/bootloader-0.9.30)
   Compiling rustc-std-workspace-core v1.99.0 (/home/mythril/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/rustc-std-workspace-core)
   Compiling zero v0.1.2
   Compiling volatile v0.4.4
   Compiling bit_field v0.10.1
   Compiling usize_conversions v0.2.0
   Compiling rlibc v1.0.0
   Compiling fixedvec v0.2.4
   Compiling xmas-elf v0.6.2
error: linking with `rust-lld` failed: exit status: 1
  |
  = note:  "rust-lld" "-flavor" "gnu" "--script=linker.ld" "--gc-sections" "/tmp/rustcAlpUTp/symbols.o" "<4 object files omitted>" "--as-needed" "-Bstatic" "/tmp/blog_os/target/bootimage/bootloader/x86_64-bootloader/release/deps/{libxmas_elf-5d65d2bf96cae931.rlib,libzero-7687372b2590b98d.rlib,libx86_64-51db3ea7fbab3b39.rlib,libvolatile-a65ebc85f9994f32.rlib,libbitflags-d0b0465c0016cac0.rlib,libbit_field-d84f3e3f48fe7d8b.rlib,libusize_conversions-83f8cb4f56f96380.rlib,libfixedvec-c637a26df226bd98.rlib,libbootloader-64a2dbd095894c20.rlib,librlibc-9db6015d2d748be6.rlib,librustc_std_workspace_core-0f32d5b194ffc9d8.rlib,libcore-df0460c8357f8b8d.rlib,libcompiler_builtins-0049d8cc3182492c.rlib}.rlib" "-L" "/tmp/rustcAlpUTp/raw-dylibs" "-Bdynamic" "--eh-frame-hdr" "-z" "noexecstack" "-L" "/tmp/blog_os/target/bootimage/bootloader/x86_64-bootloader/release/build/bootloader-f2b2226040fafd3d/out" "-o" "/tmp/blog_os/target/bootimage/bootloader/x86_64-bootloader/release/deps/bootloader-fab7ca5ce4af8903" "--gc-sections" "-O1"
  = note: some arguments are omitted. use `--verbose` to show all linker arguments
  = note: rust-lld: error: undefined symbol: __rustc::rust_begin_unwind
          >>> referenced by panicking.rs:75 (src/panicking.rs:75)
          >>>               core-df0460c8357f8b8d.core.c6d2aabfa5929589-cgu.12.rcgu.o:(core::panicking::panic_fmt::h4b21e9bee20e07ce) in archive /tmp/blog_os/target/bootimage/bootloader/x86_64-bootloader/release/deps/libcore-df0460c8357f8b8d.rlib
          

error: could not compile `bootloader` (bin "bootloader") due to 1 previous error
Error: Bootloader build failed.
Stderr: 
```
Stderr is actually empty, I did not remove it.

Here are the versions of everything:
```
bootimage 0.10.3
cargo 1.87.0-nightly (6cf826701 2025-03-14)
rustup 1.28.1 (f9edccde0 2025-03-05)
  info: This is the version for the rustup toolchain manager, not the rustc compiler.
  info: The currently active `rustc` version is `rustc 1.87.0-nightly (75530e9f7 2025-03-18)`
info: component 'llvm-tools' for target 'x86_64-unknown-linux-gnu' is up to date

```

What I have tried:
-  Building any post that uses bootimage
- ```cargo clean```
- updating cargo
- crying
- posting on [phil-opp/blog_os](https://github.com/phil-opp/blog_os/issues/1402) but was directed to post here.

Freax13, 2025-03-20:
For a temporary workaround try pinning your nightly compiler to an older version. 2025-03-17 and older should work.

Freax13, 2025-03-20:
searched nightlies: from nightly-2025-03-17 to nightly-2025-03-20
regressed nightly: nightly-2025-03-19
searched commit range: https://github.com/rust-lang/rust/compare/43a2e9d2c72db101f5fedac8b3acb78981b06bf2...75530e9f72a1990ed2305e16fd51d02f47048f12
regressed commit: https://github.com/rust-lang/rust/commit/493c38ba371929579fe136df26eccd9516347c7a

<details>
<summary>bisected with <a href='https://github.com/rust-lang/cargo-bisect-rustc'>cargo-bisect-rustc</a> v0.6.8</summary>

Host triple: x86_64-unknown-linux-gnu
Reproduce with:
```bash
cargo bisect-rustc --start=2025-03-17 --end=2025-03-20 --component rust-src --component llvm-tools --target x86_64-unknown-none --script build.sh -vvv 
```
</details>

build.sh:
```shell
#!/usr/bin/env sh

nix-shell -p autoPatchelfHook --run "autoPatchelf /home/freax13/.rustup/toolchains/$RUSTUP_TOOLCHAIN/lib/rustlib/x86_64-unknown-linux-gnu/bin/rust-lld --ignore-missing='libz.so.1 libLLVM.so.20.1-rust-1.87.0-nightly libgcc_s.so.1'" > /dev/null
nix-shell -p autoPatchelfHook --run "autoPatchelf /home/freax13/.rustup/toolchains/$RUSTUP_TOOLCHAIN/lib/rustlib/x86_64-unknown-linux-gnu/bin/gcc-ld/ld.lld --ignore-missing='libz.so.1 libLLVM.so.20.1-rust-1.87.0-nightly libgcc_s.so.1'" > /dev/null
nix-shell -p autoPatchelfHook --run "autoPatchelf /home/freax13/.rustup/toolchains/$RUSTUP_TOOLCHAIN/lib/rustlib/x86_64-unknown-linux-gnu/bin/llvm-size --ignore-missing='libz.so.1 libLLVM.so.20.1-rust-1.87.0-nightly libgcc_s.so.1'" > /dev/null
nix-shell -p autoPatchelfHook --run "autoPatchelf /home/freax13/.rustup/toolchains/$RUSTUP_TOOLCHAIN/lib/rustlib/x86_64-unknown-linux-gnu/bin/llvm-objcopy --ignore-missing='libz.so.1 libLLVM.so.20.1-rust-1.87.0-nightly libgcc_s.so.1'" > /dev/null
nix-shell -p autoPatchelfHook --run "autoPatchelf /home/freax13/.rustup/toolchains/$RUSTUP_TOOLCHAIN/lib/rustlib/x86_64-unknown-linux-gnu/bin/llvm-ar --ignore-missing='libz.so.1 libLLVM.so.20.1-rust-1.87.0-nightly libgcc_s.so.1'" > /dev/null
cargo test --target x86_64-unknown-none
```

phil-opp, 2025-03-21:
I think this should be fixed by https://github.com/rust-lang/compiler-builtins/commit/cdf8cab99cfcb98982eaa230a2603f21c08b8ae2 , which was released as part of `compiler_builtins v0.1.152`. The PR to update the `compiler-builtins` crate in the Rust standard library was also approved and should be merged soon: https://github.com/rust-lang/rust/pull/138728

Freax13, 2025-03-21:
> I think this should be fixed by https://github.com/rust-lang/compiler-builtins/commit/cdf8cab99cfcb98982eaa230a2603f21c08b8ae2 , which was released as part of `compiler_builtins v0.1.152`. The PR to [...]

Doesn't that commit just fix their tests?

phil-opp, 2025-03-21:
Yes, I think you're right... Not sure where this call is coming from then..

9thIvy, 2025-03-21:
> For a temporary workaround try pinning your nightly toolchain to an older version. 2025-03-17 and older should work.

This worked! If anyone else has the issue, I fixed it by running:
```
rustup install nightly-2025-03-17
rustup override set nightly-2025-03-17
rustup component add rust-src --toolchain nightly-2025-03-17-x86_64-unknown-linux-gnu
rustup component add llvm-tools-preview
cargo clean
cargo run
```
Not sure why llvm-tools-preview had to be re-added without a prefix like x86_64-unknown-linux-gnu was, but it works.

MaskMatrix, 2025-03-22:
1.87.0-nightly (2025-03-21) got the same problem, but not working.

Freax13, 2025-03-22:
Try using an older compiler. We should have a release with the fix soon.