rust-osdev/bootloader #503: uefi source code here is outdated

ofluffydev, 2025-04-06:


ofluffydev, 2025-04-06:
uefi is way past 0.2 now and this code is a bad example

Freax13, 2025-04-06:
This repo isn't meant to be an example for the `uefi` crate, so it seems rude to say that it's a bad example when that was never our goal. The uefi crate has had a lot of breaking changes over the years, and we haven't kept up with this because there wasn't a compelling reason for us to do so.

closed 2025-04-06 by Freax13

ofluffydev, 2025-04-06:
The breaking changes were a decision made and need to be handled. Either this is behind or the UEFI crate is bad?.. It's not a rude thing, you should have issues open for those breaking changes in a fork of the repo, focused on updating the code to fit it...

> there wasn't a compelling reason for us to do so.

Countless people are opening issues on both this repository and the UEFI one wondering how to use any of it, so I disagree. this is an issue, even if it takes months to fix, even it takes me pr-ing the repo myself to fix it. Github issues are structured to help do that.

Freax13, 2025-04-06:
This repo is not a fork of the uefi repo. I'm not aware of any people asking questions on this repo on how to use the uefi crate.